andhieSetyabudi/BQ25896 #1: BQ25896

ngohoan98, 2019-12-18:
I am a student just beginning to learn about programming. I am stuck in the library of BQ25892 for stm32f103c8t6, I don't know where to start. Can you help me?

andhieSetyabudi, 2019-12-18:
hello @ngohoan98 
can i help you ?

ngohoan98, 2019-12-18:
I am a student just beginning to learn about programming. I am stuck in the library of BQ25892 for stm32f103c8t6, I don't know where to start. Can you help me?

From: andhieSetyabudi
Sent: Wednesday, December 18, 2019 5:34 PM
To: andhieSetyabudi/BQ25896
Cc: Ngô Duy Hoàn; Mention
Subject: Re: [andhieSetyabudi/BQ25896] Fbgh (#1)

hello @ngohoan98
can i help you ?

andhieSetyabudi, 2019-12-18:
are you using bluepill? 
BQ25896 is battery-management system ( charging ) that use I2C for communication line.
so u can wiring sda & scl from STM32F103 to BQ chip. 

create the obj by calling

BQ25896  battery_charging(Wire);

to update properties of your measurement, use

battery_charging.properties(); 

or you can try to type a program, same as in readme

asenelupinn, 2019-12-18:
thank you so much

closed 2019-12-18 by asenelupinn